bschwind/key-ripper #20: Fix EMC issues, and possibly warping as a side-effect

aw, 2022-08-20:
Hi Brian, perhaps I can suggest switching to a 4-layer PCB and make the copper pours on the inner layers (GND/+3.3V), and keep the traces on the top/bottom layers.

You might have a high noise/EMI problem:

1. There are many slots in the ground planes which create "islands" and cause problems for the current return path (where would the return current flow?).
2. The signals on the bottom layer cross the slots in the top layer (and vice-versa), which is basically creating multiple dipole antennas which will radiate. See below: the red trace (F.cu) goes right through an opening (slot) in the ground plane (B.cu), and the green trace (B.cu) goes right through an opening (slot) in the other ground plane (F.cu).

[image]

In those areas where the green and red traces cross, you literally have no ground.. solving this might also fix your warping issues in #18 because you would then end up with nice clean uninterrupted ground/+3.3V planes, and much more balanced copper pours.

(these are just assumptions by the way, forgive me if i'm wrong) 😆

aw, 2022-08-20:
Just to give an idea, here's what it would look like with the ground pour in a middle layer (4-layer), with signals on top and bottom:

[image]

Notice the lack of slots and thermal relief around the SMD components?

aw, 2022-08-20:
And finally, with the +3.3V pour in the other middle layer, you can now eliminate all these long +3.3V traces (highlighted below) and simply drop some nearby vias to connect to that layer:

[image]

bschwind, 2022-08-20:
Hey @aw thanks for reporting in!

This was one of the first decisions I struggled with, until I looked at JLCPCB quotes.

For a board this size, 4 layers is significantly more expensive than a 2 layer board:

4 Layer
* 285.7mm x 116.7mm, 4 Layer, 1.6 thickness, black soldermask, lead-free HASL, 1 oz copper: 10137 yen + shipping

2 Layer
* 285.7mm x 116.7mm, 2 Layer, 1.6 thickness, black soldermask, lead-free HASL, 1 oz copper: 2632 yen + shipping

I also get an extra slight penalty for my choice of black soldermask and lead-free HASL for the 4 layer board:
[image]

One of the goals is to be able to produce this cheaply, so I opted to try to fit things in 2 layers to see if it's feasible. Of course 4 layers would be much easier to route, but I also feel like my routing skills improved after this project a bit :)

I won't try to claim EMI/noise isn't a problem with this board, as I don't have the equipment to make that claim. I believe the highest speed signal on the board is the USB line, running at most USB Full Speed (12Mbps). I did try to make sure I had an uninterrupted ground plane underneath those data lines, and tried to provide short ground return paths where I had to interrupt it, as shown:

[image]

For the majority of the rest of the lines in the board, they are used as such:

* One by one, each column on the board is set high via a GPIO pin. The MCU then sleeps a bit to make sure the signal has propagated.
* All GPIO row pins are read, to see which are high, which determines key presses for those rows/columns

In that sense, I'm not worried about the signals interfering with each other, but I understand how an interrupted ground can cause radiation issues. From what I understand, it's related to the rise time of the signal and not the frequency.

So could I possibly attach small valued capacitors or something similar which would reduce the rise time?

Otherwise, the board functions fine, but I unfortunately don't have the equipment to measure EMI (let me know if you do, or know someone who does though!).

Just like code performance, I'd prefer to measure a before and after before applying an optimization which may not make an appreciable difference on the end product.

Regarding PCB warping, that issue is more or less solved with v0.2. It's a tiny bit curved, but when combined with a top plate, or screwed into a case, that curve will go away. I could experiment with a hatched ground pour instead of a solid one to see if that improves it further.

aw, 2022-08-20:
> 4 Layer
> 
>     * 285.7mm x 116.7mm, 4 Layer, 1.6 thickness, black soldermask, lead-free HASL, 1 oz copper: 10137 yen + shipping
> 
> 
> 2 Layer
> 
>     * 285.7mm x 116.7mm, 2 Layer, 1.6 thickness, black soldermask, lead-free HASL, 1 oz copper: 2632 yen + shipping

Haha yeah I think that's a pretty big price difference. From an electrical standpoint if you've already confirmed it "works fine" and doesn't have any issues then great. That's what prototypes are for anyways!

I also don't have EMI test equipment but maybe I should? hmmmm...

bschwind, 2022-08-20:
I've been using it as my main keyboard for a month or so now, it works well! There are some usb descriptor issues I need to sort out on macos but otherwise it's stable and totally usable!

Can't claim no EMI noise though, as I said. I'd be super curious to test it out.

closed 2022-08-27 by aw