bschwind/key-ripper #27: Case Construction

bschwind, 2022-10-29:
I'd like to create a case for this keyboard, with two varieties:

* A bottom-plate case, to protect the electronics on the bottom of the PCB. It will still be open from the sides
* A 3D-printable (and maybe eventually aluminum CNC-able) case

## Measurements

* Currently the tallest component on the board is the USB-C connector, at 3.26mm, followed by the QSPI flash module at 1.95mm, and the kailh hotswap sockets (if they're used) at 1.85mm.

USB-C Connector
[image]
Flash Memory
[image]
Kailh Hotswap Socket
[image]

## Constraints

* The USB-C connector should still be mounted on the same side as the electronics for easier manufacturing
* The reset and USB-boot buttons should ideally still be accessible in some way (side-mounted buttons? holes in the bottom plate?)
* The design should require as few screws as possible, no nuts, and especially no custom 3D-printed spacers as I've used before.

## Bottom Plate Design

Following on the design of other keyboards (such as the [7spro](https://salicylic-acid3.hatenablog.com/entry/7spro-build-guide)), there will be M2 spacers which go _through_ the main PCB, with screws securing it to the top and bottom plates.

[image]

I would try to get away with 8mm spacers first if the USB connector isn't as tall as it says in the datasheet, otherwise 9mm spacers will have to do.

The screws should be as flat as possible to stay flush with the top and bottom plate. 3-4mm length should be fine.

[image]
[image]

- [ ] To be determined - The drill radius needed to allow M2 spacers to pass through the PCB.

bschwind, 2022-11-12:
Cleanup tasks:

- [ ] Move row 0 diodes (top row or F row) below their switches instead of above to keep them off the edge of the PCB
- [ ] Move row 0 traces to be below the switch

skywhale, 2022-11-28:
If you are okay with an aluminum back plate, then using rivet nuts ([an example](https://www.monotaro.com/p/0674/4832)) is one easy way to make custom feet attachable.

bschwind, 2022-11-29:
Here's a good video showing how a non-hotswap keyboard is expected to be constructed in a case similar to what I want to build:

https://youtu.be/017aGAJiNJg

bschwind, 2022-11-30:
Some findings from @mcginty for adding threading to the PCB:

https://www.keyelco.com/category.cfm/Spacers-Standoffs/Standoffs-Spacers-Inserts-Surface-Mount/id/1335

https://www.mouser.com/ProductDetail/Keystone-Electronics/24929?qs=Znm5pLBrcAJPd1Bwfdw1vA%3D%3D

These could be soldered directly to the PCB as a solid, no-fuss solution to adding threading points on the PCB.

Very similar to @skywhale 's suggestion of rivet nuts, but without the need of a special tool (besides a soldering iron) for attaching it.

bschwind, 2022-12-04:
Progress with the "PCB with threaded inserts" version of the case:

[image]

bschwind, 2023-01-27:
More progress!

[image]
[image]

closed 2025-07-03 by bschwind