bschwind/key-ripper #41: ESD protection on the USB connector

bschwind, 2023-01-09:
I've heard the part USBLC6-2SC6 is commonly used for ESD protection on the USB data lines.

skywhale, 2024-02-05:
It happened with `lite-matsukazeso` today. When Alisaun touched the volume dial, the panel controller became non-responsive and it triggered a software restart.

```
Feb 05 10:32:27 lite-matsukazeso-portal tonari[888668]: got a serial message: DialValue { diff: 1 }
Feb 05 10:32:27 lite-matsukazeso-portal tonari[888668]: Handling hardware action DialTurns { ticks: 1 }.
Feb 05 10:32:27 lite-matsukazeso-portal tonari[888668]: resetting the audio volume.
Feb 05 10:32:27 lite-matsukazeso-portal tonari[888668]: Share screen audio volume set to: -18 dB
Feb 05 10:32:27 lite-matsukazeso-portal tonari[888668]: portal state updated: PortalState { mode: Normal(NormalState { share_screen_menu_state: NotShown, share_screen_volume: 11 }), casting_state: CastingState { sending_start: None, receiving_start: Some(Timestamp(1707096691.974275954s)) }, current_peer: 10.42.191.39, debug_overlay: false, local_maintenance_state_preference: Nominal, remote_maintenance_state_preference: Nominal }
Feb 05 10:32:27 lite-matsukazeso-portal tonari[888668]: [main] PanelActor handle error: Operation timed out, shutting down.
Feb 05 10:32:27 lite-matsukazeso-portal tonari[888668]: Thread [PanelActor] shutting down the actor system
Feb 05 10:32:27 lite-matsukazeso-portal tonari[888668]: Thread [PanelActor] setting the system_state value to ShuttingDown
Feb 05 10:32:27 lite-matsukazeso-portal tonari[888668]: [main] system shutting down.
Feb 05 10:32:27 lite-matsukazeso-portal tonari[888668]: [main] calling pre-shutdown callback.
```

bschwind, 2024-07-23:
The part I chose wasn't great in terms of availability. This part should be easier to source:

USBLC6-2SC6

https://jlcpcb.com/partdetail/Stmicroelectronics-USBLC62SC6/C7519

PR #66 (2024-07-21): ESD Protection
Closes #41
closed 2024-07-23 by bschwind