bschwind/key-ripper #55: Add case images and design files, if there are any

DeflateAwning, 2023-11-18:
I'm assuming you also had a 3d-printed case or similar? You should post info about it! If not, my mistake!

bschwind, 2023-11-18:
Yes, I did make a case for it! In fact it's still a work in progress which I recently picked up again.

This is quite obscure, but the code for it is currently in this repository:

https://github.com/bschwind/opencascade-rs/blob/keyboard-case-v2/examples/src/keyboard_case.rs

I actually started that project just to be able to define the case in Rust code, but it has turned into its own project which I also want to flesh out.

Here are some previews of it:

[image]

[image]

Once the code in opencascade-rs gets further along and more mature, I'll move the code for the keyboard case into this repo. Currently I often need new features added to opencascade-rs, so I add those alongside development of the keyboard case.

DeflateAwning, 2023-11-19:
Very neat project, very awesome!

I'm sure you know about OpenSCAD, but check out BOSL2 for OpenSCAD. It has a really nice API (e.g., anchor argument for zcyl, etc.), and could potentially be used as a model as least somewhat for the opencascade-rs project!

bschwind, 2023-11-19:
Could you explain anchor arguments a bit more and what you like about the API? I think I kind of understand them.

For positioning right now my library mostly uses workplanes, which you can construct directly on a typical plane like XY, or they can be constructed from the face of an object.

I still need to figure out a good API for attaching objects together with constraints though.

PR #68 (2025-07-03): Add case STEP files
Closes #55
closed 2025-07-03 by bschwind